MajicDesigns/MD_MIDIFile #3: Can't compile MD_MIDIFile_Dump

vintageradio, 2017-10-07:
I get the following error when I try to compile this example.  All of your other examples that I tried work fine
(loop, play, dump_raw):
`Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\john\AppData\Local\Temp\ccbxBadg.ltrans0.ltrans.o: In function `loop':

C:\Users\john\Documents\Arduino\libraries\MD_MIDIFile\examples\MD_MIDIFile_Dump/MD_MIDIFile_Dump.ino:88: undefined reference to `MD_MIDIFile::dump()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.
`

MajicDesigns, 2017-10-07:
Please read the comments at the top of the program file
`// MIDIFile.h library settings must be set to values below before compiling
// #define	DUMP_DATA			1
// #define	SHOW_UNUSED_META	1
//
// REMEMBER TO TURN ON LINE ENDING IN SERIAL MONITOR TO NEWLINE
//
`

closed 2017-10-07 by MajicDesigns

vintageradio, 2017-10-07:
Dumb rookie mistake. Thanks so much!