MajicDesigns/MD_MIDIFile #4: Example program stops reading from SD after 60 seconds

erniberni, 2018-05-09:
Hi,
from the examples I tried  MD-MIDIFile_Loop and MD-MIDIFile_Play. Both programs play the midi files for approximately 60s then reading from the SD card stops and the serial debug shows the line
 M T2:  Ch 2 Data 80 60 0
every 8 to 9 seconds. After a while reading and playing starts again and after a while stops again.
I am using the SD card reader of an Ethernet Shield. I tried two different SD cards and Arduino Uno and Mega with same behaviour.

Reagrds , Reinhard

MajicDesigns, 2018-05-09:
Which .MID file was it playing?

erniberni, 2018-05-10:
I looped the TWINKLE.MID from the SD_Card_Files directory. But same behaviour also with other files.

MajicDesigns, 2018-05-11:
Both the defines 
#define	DUMP_DATA
#define	SHOW_UNUSED_META 
in MD_MIDIFile.h need to be set to 0. Looks like the UNUSED_META was set to 1 on the release version. Will re-release once all tested ok. In the meantime you can change your version.

closed 2018-05-11 by MajicDesigns

erniberni, 2018-05-11:
UNUSED_META was set to 1, I changed to 0. Now it's working.
Thank you very much.