MajicDesigns/MD_MIDIFile #5: Midi Track Functions

cpethica, 2018-07-01:
Hi! 

Thanks for the great code, I've used it once already to play MIDI for a solenoid powered instrument:

https://github.com/cpethica/Cymbal-Tree

I am now working on a project with some Indonesian artists to remake the system to play a gamelan instrument. Rather than just playback though I would like the piece to be more interactive and am building a MIDI controller that allows people to control aspects of the playback. With this in mind I'm trying to find a way to mute or control the volume of specific MIDI tracks within a file (probably through MIDI input from another arduino - or button inputs directly controlling your code). This would allow pre-composed loops and sequences to be built up or controlled by the user.

Do you think this is something that would be possible with your software? Any tips and pointers for good places to start would be really appreciated!

Thanks

Charlie

MajicDesigns, 2018-07-01:
It should be possible. Probably the easiest way is for the structure passed through the callback to include the trackID. This will let the application decide whether to ignore the track and will also keep the track playing so that when/if it is turned back on it is still in synch with the rest of the tracks.

The recent addition of the meta event callback has this and it is probably a good thing to put into both the midi and sysex events as well. I can make the change or you can do it and resubmit for inclusion in the library code.

MajicDesigns, 2018-07-01:
Actually, just checked the code and trackid is already there. So you just need to ignore any midi events that come from the track you want to ignore and you should be ok.

closed 2018-07-01 by MajicDesigns