MajicDesigns/MD_MIDIFile #8: Some MIDI files play with corruption, then fine the second time

thorr2, 2019-07-21:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library? Yes
[ ] Have you checked this has not already been submitted and/or resolved? Yes
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)

## Certain MIDI files have issues
A bunch of MIDI files have problems playing back correctly.  They skip through a bunch of the song super fast and eventually start playing normally part way through the song.  Playing them again in a loop sometimes plays them correctly.  

## Your Environment
**Library Version:**  Latest
**Arduino IDE version:** Latest
**Hardware model/type:** Mega 2560
**OS and Version:**  Latest

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.

Download the KQ5 GM MIDI files from here, and try to play in a loop GM/08 - Bandit Camp.MID (after renaming it to BANDIT.MID)

http://www.mirsoft.info/wogm_download.php?data=YTozOntpOjA7czo0OiIxOTYzIjtpOjE7aToxNTYzNzQ3MjQ3O2k6MjtpOjM7fQ==

The first time it is played, it plays super fast and skips a bunch of it, then plays fine the second time.  A lot of other files in the zip file have issues as well.  Plenty of other MIDI files that I tried have no issues.

## Expected Behaviour
Explain what should happen.
It should just play them correctly every time, not just sometimes.

## Actual Behaviour
Explain what happens instead. Provide log messages if relevant.

## Code Demonstrating the Issue
````
Insert your code here.
See examples folder.  I tried MD_MIDIFILE_PLAY and MD_MIDIFILE_LOOP.  I am using Serial1 on the 2560, and changed the code in the source files to from Serial.xxx to Serial1.xxx.  The CS is on pin 53.  I changed the maximum midi channels to 32.
````

Thank you for your help.

MajicDesigns, 2019-07-22:
I have tried it on my setup (Arduino Uno) and I don't get the same result as you.

This usually means that there is a problem related to the hardware target compiler (MEGA in your case) behaving differently from the one I test on (Uno).

Have you got an Uno that you can try it on to confirm this?

thorr2, 2019-07-22:
Hello Marco,

Unfortunately, no.  I only have a Mega.  I needed to do this on a Mega for
my project because I needed more than one serial port.

It is strange because it is consistent and random at the same time.  I have
a GS Reset MIDI file I have been playing before other midi files, and if I
put three of them before the Bandit Camp midi file, it plays fine every
time as long as I don't change the order of the file playback.  If I do
things in a different order or quantity of GS Reset files, then it gets
messed up.

Thank you very much for your help.  I plan to donate a bit for your project
because I appreciate what you put together so far and for your
help/feedback now.

Michael Hart

MajicDesigns, 2019-07-23:
Thanks for the donation. I have ordered a MEGA and we'll see how we go with that once it gets here from China.

thorr2, 2019-07-23:
Terrific!  Thanks Marco.

thorr2, 2019-08-11:
Hi Marco, I think I found a new clue to what is going on with the Bandit Camp midi file.  If I tell the program to play a new file and specify the same Bandit Camp midi file every time while it is already playing it, it will play it at super high speed up to the point in the song where it was before I sent it a new file to play, and then play it at normal speed from that point forward for the remainder of the song.  If I play a different file, the different file plays just fine as long as it is not one of the bad files like Bandit Camp.  If I then play Bandit Camp again, it fast forwards through the whole song and plays it again normally.  So it seems like there is some kind of song position indicator that is not getting reset when playing Bandit Camp as a new song.  If I knew what variable to reset and how, that might be a workaround to add a line in there to reset that variable before playing the next file.  

Thanks,
Michael

MajicDesigns, 2019-08-11:
Can you document what exactly you are doing to cause this problem.
1. What sketch are you using?
2. What songs are on the SD card?
3. What sequence of keys/operations are you doing that causes this problem? (in detail)

thorr2, 2019-08-12:
I modified one of your sketches and the libraries so that I was using Serial 1 for the MIDI, and I am using the Arduino Serial Monitor on Serial to type in the path to the MIDI file.  See the code below.

On the SD card, I have the files: 
MIDI/GS_RESET.MID   (a silent midi file to start with on power up that should reset everything but it does not stop stuck notes, so I also added a midiSilence in the setup)
PROGRAMS/KQ5/GM/08BANDIT.MID (and others in the same directory renamed from the zip file)

Open the Serial Monitor, change it to 9600 baud and Newline.  Type in PROGRAMS/KQ/GM/08BANDIT.MID and hit Enter.  You can play other files the same way.  If you keep typing PROGRAMS/KQ/GM/08BANDIT.MID with Enter, it will fast forward to the same point in the song where you hit enter the last time and just a bit after that, and then play it at normal speed when it gets there.

I start the program playing a GS_RESET.MID file so it will loop that in silence until it gets the next file name.

Here is the code (remember to change the libraries to Serial1 for MIDI and wire it up that way on the board).  You may also be able to leave it at Serial since you are inputting to it via the Serial Monitor, and it is outputting to the MIDI but I did not test this.

A ton of other midi files play fine, but this group has issues and I occasionally come across others like it.  Thanks for your help!

Michael

// Play a file from the SD card in looping mode, from the SD card.
// Enter PATH/TO/FILENAME.MID via the Serial Monitor to interrupt the song and play the next song.
//
// Uses Mega 2560 with Serial1 for MIDI output and Serial for Filename Input.  Be sure to change the library files to Serial1 for MIDI.
// When using the Serial Monitor, be sure to set it to use New Line only in the drop down.
//
// Hardware required:
//  SD card interface - change SD_SELECT for SPI comms

#include <SdFat.h>
#include <MD_MIDIFile.h>

#define USE_MIDI  1

#if USE_MIDI // set up for direct MIDI serial output on Serial1
   #define DEBUGS(s)
   #define DEBUG(s, x)
   #define DEBUGX(s, x)
   #define SERIAL_RATE 31250
#else // don't use MIDI to allow printing debug statements
   #define DEBUGS(s)     Serial1.print(s)
   #define DEBUG(s, x)   { Serial1.print(F(s)); Serial1.print(x); }
   #define DEBUGX(s, x)  { Serial1.print(F(s)); Serial1.print(x, HEX); }
   #define SERIAL_RATE 57600
#endif // USE_MIDI

// SD chip select pin for SPI comms.
// Arduino Ethernet shield, pin 4.
// Default SD chip select is the SPI SS pin (10).
// Other hardware will be different as documented for that hardware.
#define  SD_SELECT  53                 // On Mega 2560

#define ARRAY_SIZE(a) (sizeof(a)/sizeof(a[0]))

// The files in the tune list should be located on the SD card 
// or an error will occur opening the file and the next in the 
// list will be opened (skips errors).
char *loopfile = "MIDI/GS_RESET.MID";  // loopfile is the name of the MIDI file including the path.  Start with GS_RESET.MID so it is silent.
const byte numChars = 32;              // used for the maximum length of the path and file name coming in via Serial
char receivedChars[numChars];          // a character array to store the received filename as it is coming in the Serial port
boolean newData = false;               // Did we get a complete new filename via the Serial port?

SdFat SD;
MD_MIDIFile SMF;

void midiCallback(midi_event *pev)
// Called by the MIDIFile library when a file event needs to be processed
// thru the midi communications interface.
// This callback is set up in the setup() function.
{
#if USE_MIDI
  if ((pev->data[0] >= 0x80) && (pev->data[0] <= 0xe0))
  {
    Serial1.write(pev->data[0] | pev->channel);
[... 95 lines cut ...]
  SMF.setMidiHandler(midiCallback);
  SMF.looping(true);
  midiSilence();                   // Silence any hanging notes

  // use the next file name and play it
  DEBUG("\nFile: ", loopfile);
  SMF.setFilename(loopfile);
  err = SMF.load();
  if (err != -1)
  {
    DEBUG("\nSMF load Error ", err);
    while (true);
  }
}

void loop(void)
{
  // play the file
  if (!SMF.isEOF())
  {
    SMF.getNextEvent();   // Play MIDI data
    recvWithEndMarker();  // Get a Serial character if any comes in and construct a new file name over time
    NewMidiFile();        // Change the file name if a new filename came in
  }
}

MajicDesigns, 2019-08-25:
I have spent 1.5 days trying to replicate your issue and I just can't get the same effect of speeded up 'catch up' playback. I have created a new example (Player_CLI) that does similar to your application. Maybe you can try using this to see if you still get the problem. It is in the newly uploaded  v2.3.1

thorr2, 2019-08-25:
Thank you Marco!  I won't have time to try it until next weekend because I
will be traveling.  I am looking forward to it.  Thank you for your
efforts!  I will keep you posted.

thorr2, 2019-09-01:
Hi Marco, thanks again for your efforts.  I tried the new version with the new Player_CLI program and unfortunately got the same results.  When you were testing, were you using the Bandit Camp GM file that I linked previously?  Many other files work fine.  With the new Player_CLI program, I showed the output of the MIDI stream in the console, and it was printing as fast as it could while it was in fast forward mode and this actually slowed it down compared to not displaying that output.  

Also, I am thinking about getting a Teensy 3.5 with the built-in micro SD slot if I can't get the 2560 I have to work reliably.   It is much smaller and more capable than the Mega2560.  I know that the SDFAT library has special code for the Teensy 3.5/3.6 (it is not defined by a CS pin and there is an example in the SDFat library).  Do you foresee or know of any major issues adapting your code to use the micro SD slot on the Teensy?   

Thanks again,
Mike

thorr2, 2019-09-01:
Hi Marco,

I just figured out the difference!  If I put the MIDI file in the root directory, it plays fine.  I was playing the file from PROGRAMS/KQ5/GM/08BANDIT.MID and it was having the speed up problem.  I just tested with the file in both locations and it always has issues in the folder, and never has issues in the root folder.  Can you try this out and let me know what happens?

Thanks,
Mike

MajicDesigns, 2019-09-01:
Won’t be able to do that for a few days. You could check for buffer overruns with a name that long,  which could cause vale’s following the string to get clobbered.

thorr2, 2019-09-02:
I am not sure how to do that, but it plays the correct file so at least that works with the long name.  I can happily wait until you have time to take a look as I am busy anyway with another project I am working on in parallel.  Thanks again!

MajicDesigns, 2019-09-14:
As expected this was a buffer overflow. The file name was being copied into a space for 8.3 (12 characters) and that was scribbling all over other memory. setFilename() no longer allows the buffer to overflow, so this bug is quashed. I have added another method setFileFolder() which enables changing the current working folder to a new location (PROGRAMS/KQ5/GM/ in your example). Subsequent 8.3 file references are within this folder.

Released as version 2.3.2

thorr2, 2019-09-16:
Awesome!  Thank you so much Marco!  I am looking forward to testing this
out!

closed 2019-12-30 by MajicDesigns