MajicDesigns/MD_MIDIFile #9: Hardware Interface wiring

the-force-2000, 2019-08-01:
## Subject of the issue
First off, I can't thank you enough for putting this together.   I'm just starting my project and looking forward to using this library.    

I was looking at your Hardware Interface documentation and I'm suspicious about the Midi connector wiring.   Everything I have been able to find shows the Midi Out pin 4 going to +5 (through pull up resistor), and pin 5 going to Tx, but you show them swapped.   Am I missing something?   Even your linked reference shows the pins swapped from your diagram.   I only need to drive the output so not looking at the Midi In wiring.   Perhaps it doesn't matter?

MajicDesigns, 2019-08-01:
It matters. The note in the circuit diagram ("MIDI connectors are viewed from the back (or wiring) side") tries to make this clear as I could not find any source that would give me a definitive numbering for the pins on the MIDI sockets. If you view the socket connector from the front (plug in) then pins 4 and 5 are swapped (ie the perspective matters).

closed 2019-08-01 by MajicDesigns

the-force-2000, 2019-08-01:
Thanks!!    I’m shocked that there isn’t standardized pin numbering for these connectors.

Thank you.

MajicDesigns, 2019-08-01:
It surprised me too, but maybe it was just me not finding one. Usually the pins (or at least pin 1) are numbered on the connectors but the ones I had did not seem to identify them in any way.

the-force-2000, 2019-08-05:
Sorry if this is a dumb question.  I’m new to MIDI.   I see that most keyboards now use USB for MIDI and don’t even seem to have the MIDI plugs.   Is the MIDI plug even necessary to use with the MIDIFile library?   Could I just use the Arduino serial port?  I realize this would be a pain because I couldn’t use it at the same time for serial loading and debugging, but wondering if the plug is necessary.

Thanks,
-steve.

MajicDesigns, 2019-08-05:
You can indeed just use the serial port with a serial to usb converter. I use the midi plugs to interface to a midi keyboard and the serial port for a midi synthesizer.

the-force-2000, 2019-08-05:
Thanks, but I’m still confused.   The Arduino has a USB on it, so not sure what you mean by “serial to USB converter”.   Can’t I just connect the Arduino USB directly to a USB MIDI synthesizer ?

Thanks,
-steve.

the-force-2000, 2019-08-05:
Never mind – I found some info on line that indicates that the native Arduino USB serial cannot be used without modifying the firmware.  I’d rather not to that so I’ll stick with your design and get a MIDI socket.   Thanks just the same.   I’ll let you know how the project goes (have to finish last project first, so may be a while) and shoot you a donation if all goes well!

Thanks,
-Steve.

From: Swaine (US), Steve
Sent: Monday, August 05, 2019 6:45 AM
To: MajicDesigns/MD_MIDIFile <reply@reply.github.com>
Subject: Re: [MajicDesigns/MD_MIDIFile] Hardware Interface wiring (#9)

Thanks, but I’m still confused.   The Arduino has a USB on it, so not sure what you mean by “serial to USB converter”.   Can’t I just connect the Arduino USB directly to a USB MIDI synthesizer ?
Thanks,
-steve.

the-force-2000, 2019-08-13:
Thanks again.   Worked great!    Just sent you a donation from my PayPal account.

Thanks,
-steve.

MajicDesigns, 2019-08-13:
Thanks very much for your contribution!