MajicDesigns/MD_MIDIFile #13: Some files which played under the previous version now fail to play with error 7.

urbantigerau, 2020-03-25:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library? Yes.
[ ] Have you checked this has not already been submitted and/or resolved? If so I cannot find it
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/) No this is highly MD_MIDIFile specific

## Subject of the issue
Files which did play in the previous version now generate an error 7: Too Many Tracks

## Your Environment
**Library Version:** Current
**Arduino IDE version:** 1.8.7
**Hardware model/type:** Arduino Mega with 1280 processor
**OS and Version:** Win10 Home, 1909  latest patches.

## Steps to Reproduce
Explain how to reproduce this issue. Please provide working code below to demonstrate the issue.
Using the built in CLI example try to play the files in the attached zip. In the previous release they would play, but error 7, now.
## Expected Behaviour
Files used to play, 

## Actual Behaviour
Files error with error number 7, Too Many Tracks

## Code Demonstrating the Issue
Use the CLI example code
Insert your code here.
I have added no code.
[Generate_error_7.zip](https://github.com/MajicDesigns/MD_MIDIFile/files/4380774/Generate_error_7.zip)

MajicDesigns, 2020-03-25:
Change the number of tracks in the library header file to suit your application. The more tracks the less memory you have available for other things.

The parameter to change is in the MIDI_MAX_TRACKS define MD_MIDIFile.h file
`#define MIDI_MAX_TRACKS 16`

closed 2020-03-25 by MajicDesigns

urbantigerau, 2020-03-25:
Of course, many thanks, oddly I didn’t change this previously. I don’t change things I don’t understand properly as a rule. Regards, Fred

MajicDesigns, 2020-03-25:
It was set to 24 in the previous release but I had a number of people who could not open files because of insufficient RAM. It was 16 in the version before that. Most MIDI files have 16 or less tracks, as this was a limit in the original specification.

urbantigerau, 2020-03-26:
Thanks again, the more I think about it I recall something in the documentation about that RAM trade-off. It had slipped my mind and I will look again. Considerable distraction here too, with COVID19 stuff, contributing to absent mindedness. Stay safe, Regards.