MajicDesigns/MD_MIDIFile #14: Files in doxygen documentation give a 404 error

SteveAboveTheClouds, 2020-08-08:
I'm not a doxygen expert, but the links under File List and File Members on this page:

https://majicdesigns.github.io/MD_MIDIFile/files.html

all give 404 errors.

Just seems like something you would want to fix.

MajicDesigns, 2020-08-08:
Thanks for letting me know. It is something that I would want to fix.

Unfortunately, the link in your message seems to work fine and a direct access to the online docs "File Members" (https://majicdesigns.github.io/MD_MIDIFile/globals.html) seem to work for me as well.

SteveAboveTheClouds, 2020-08-08:
The links to that level work, but if you try to link to a file, for example:

https://majicdesigns.github.io/MD_MIDIFile/files.html  This looks ok

But click on  MD_MIDIFile.cpp on the right hand side to see the source
code, the link is:
https://majicdesigns.github.io/MD_MIDIFile/_m_d___m_i_d_i_file_8cpp.html

THAT goes to a 404 error.

All of the *right hand links* for File Members and File List behave the
same way on my computer (Windows 10 & Chrome).  All other parts of the
doxygen presentation seem to work as expected.

Best regards,
Steve Leininger

MajicDesigns, 2020-08-09:
OK, thanks for the clarification. The files definitely don't appear when viewed from Github, but local copies in the docs folder of the library seem to work ok. Do you find the same thing?

SteveAboveTheClouds, 2020-08-09:
Yes, the files in the docs directory look as one would expect them to look.

I can't directly access the gethub.io mapping you would have to use for the
doxygen presentation.  It seemed odd that files weren't showing up.

MajicDesigns, 2020-08-09:
The files are there, I just checked, and they are uploaded from my machine. I'll need to investigate this further.

MajicDesigns, 2020-08-09:
The solution is found at https://github.blog/2009-12-29-bypassing-jekyll-on-github-pages/ and works when implemented.

Looks like all the other repositories are going to have the same issue.

closed 2020-08-09 by MajicDesigns