MajicDesigns/MD_MIDIFile #15: Compiling Error

PaulderGraph, 2020-11-22:
Hello, I would like to use this library but I get an Compiling error. I use the latest version of MD_MIDIFile and the latest of the SdFat Library.
I compiled for Arduino Uno and Micro but I dont think it has to do with the Platform.
This is the error I get:
[image]
Thanks in Advance
Paul

MajicDesigns, 2020-11-22:
Can you please cut and paste the error from the compiler output as I cannot read the graphic (too low resolution).

MajicDesigns, 2020-11-22:
Also, try using an earlier version of the SDFat library.

PaulderGraph, 2020-11-22:
Thank you for the fast response. I will definately try older Versions. Tomorrow I will send you the Error Codes. 
Thanks 
Paul

PaulderGraph, 2020-11-23:
`Arduino: 1.8.13 (Windows 10), Board: "Arduino Micro"

In file included from C:\Users\PaulGraf\Documents\Arduino\libraries\MD_MIDIFile-master\examples\MD_MIDIFile_Dump\MD_MIDIFile_Dump.ino:7:0:

C:\Users\PaulGraf\Documents\Arduino\libraries\MD_MIDIFile-master\src/MD_MIDIFile.h: In member function 'void MD_MIDIFile::setFileFolder(const char*)':

C:\Users\PaulGraf\Documents\Arduino\libraries\MD_MIDIFile-master\src/MD_MIDIFile.h:899:87: error: no matching function for call to 'SdFat32::chdir(const char*&, bool)'

   void setFileFolder(const char* apath) { if (apath != nullptr) _sd->chdir(apath, true)};

                                                                                       ^

In file included from C:\Users\PaulGraf\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatLib.h:27:0,

                 from C:\Users\PaulGraf\Documents\Arduino\libraries\SdFat-master\src/SdFat.h:34,

                 from C:\Users\PaulGraf\Documents\Arduino\libraries\MD_MIDIFile-master\examples\MD_MIDIFile_Dump\MD_MIDIFile_Dump.ino:6:

C:\Users\PaulGraf\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatVolume.h:67:8: note: candidate: bool FatVolume::chdir()

   bool chdir() {

        ^~~~~

C:\Users\PaulGraf\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatVolume.h:67:8: note:   candidate expects 0 arguments, 2 provided

C:\Users\PaulGraf\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatVolume.h:76:8: note: candidate: bool FatVolume::chdir(const char*)

   bool chdir(const char *path);

        ^~~~~

C:\Users\PaulGraf\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatVolume.h:76:8: note:   candidate expects 1 argument, 2 provided

C:\Users\PaulGraf\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatVolume.h:251:8: note: candidate: bool FatVolume::chdir(const String&)

   bool chdir(const String& path) {

        ^~~~~

C:\Users\PaulGraf\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatVolume.h:251:8: note:   candidate expects 1 argument, 2 provided

exit status 1

Error compiling for board Arduino Micro.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

PaulderGraph, 2020-11-23:
This are the error codes. Ignore if something is marked as `Code`

PaulderGraph, 2020-11-23:
I also get the error when using verion 1.0.0 of SdFat

MajicDesigns, 2020-11-23:
The issue is the changes to SdFat with the release of their version 2+. the chdir() method changed to not accept a he previoulks optional boolean second parameter. Library has been updated to work with the new SdFat. Released as v2.5.1.

closed 2020-11-23 by MajicDesigns