MajicDesigns/MD_MIDIFile #16: No Midi output use!

PaulderGraph, 2020-12-02:
Hello,
How is it possible to just use the Serial output without the midi stream?
How would the sketch look like?
What are the needed Commands?
For me its not clear out of the examples.

Thanks, Paul

MajicDesigns, 2020-12-02:
I am really not sure what you are asking. 

If you are sending the MIDI data out of the serial port, then that serial stream is the MIDI stream and you can't be sending other data in the stream or you will confuse the MIDI device at the other end. If you have more than one hardware serial ouput  (eg a MEGA processor board) or use the SoftwareSerial library,  then you split the streams and write appropriate data to each stream

Most of the examples have a #define that allows to switch the serial output to a formatted print for debugging purposes. This choice is made in the MD_MIDIFile callback function, so you should look there.

PaulderGraph, 2020-12-04:
I want a midi stream over the serial monitor with delays. I dont need anything else. I would like a basic sketch whats really needed for the lib.

MajicDesigns, 2020-12-05:
By "with delays" I assume you mean with the correct timing for the music playback.

The simplest example is MIDIFile_Loop, which plays the same file over the serial port, over and over again. You need to change the name of the file to your own in the code.

PaulderGraph, 2020-12-06:
[image]
Now I at the end of the line, there is a Question Mark and then 2 random chars.
I edited the code a bit but before it did it too

`// Play a file from the SD card in looping mode, from the SD card.
// Example program to demonstrate the use of the MIDFile library
//
// Hardware required:
//  SD card interface - change SD_SELECT for SPI comms

#include <SdFat.h>
#include <MD_MIDIFile.h>

#define USE_MIDI  0  // set to 1 for MIDI output, 0 for debug output

#define DEBUGS(s)     Serial.print(s)
#define DEBUG(s, x)   { Serial.print(F(s)); Serial.print(x); }
#define DEBUGX(s, x)  { Serial.print(F(s)); Serial.print(x, HEX); }
#define SERIAL_RATE 57600

const uint8_t SD_SELECT = 53;

#define ARRAY_SIZE(a) (sizeof(a)/sizeof(a[0]))

// The files in the tune list should be located on the SD card 
// or an error will occur opening the file and the next in the 
// list will be opened (skips errors).
const char *loopfile = "bh.mid";  // simple and short file

SdFat	SD;
MD_MIDIFile SMF;

void midiCallback(midi_event *pev)
// Called by the MIDIFile library when a file event needs to be processed
// thru the midi communications interface.
// This callback is set up in the setup() function.
{
    Serial.write(pev->data, pev->size);
  DEBUG("\nM T", pev->track);
  DEBUG(":  Ch ", pev->channel+1);
  DEBUGS(" Data");
  for (uint8_t i=0; i<pev->size; i++)
  {
    DEBUGX(" ", pev->data[i]);
  }
}

void setup(void)
{
  int  err;
  Serial.begin(SERIAL_RATE);
  DEBUGS("\n[MidiFile Looper]");
  // Initialize SD
  if (!SD.begin(SD_SELECT, SPI_FULL_SPEED))
  {
    DEBUGS("\nSD init fail!");
    while (true) ;
  }

  // Initialize MIDIFile
  SMF.begin(&SD);
  SMF.setMidiHandler(midiCallback);
  SMF.looping(true);

  // use the next file name and play it
  DEBUG("\nFile: ", loopfile);
  err = SMF.load(loopfile);
  if (err != MD_MIDIFile::E_OK)
  {
    DEBUG("\nSMF load Error ", err);
    while (true);
  }
}

void loop(void)
{
  // play the file
  if (!SMF.isEOF())
  {
    SMF.getNextEvent();
  }
}`

PaulderGraph, 2020-12-06:
Ok, found the error, it was the `Serial.write(pev->data, pev->size);`
Can I use the code without the `DEBUG();`
What is it for.

MajicDesigns, 2020-12-06:
DEBUG is for debugging. Just turn it off at the top of the file (#define).

PaulderGraph, 2020-12-06:
Yes but how can I replace it with a Serial.println();

MajicDesigns, 2020-12-06:
Serial.write() writes the data to the serial output. That is already there.

Serial.println() would be incorrect as it adds in an end of line to the stream.

PaulderGraph, 2020-12-08:
So, I got this
How can I test if its the end of the song?

MajicDesigns, 2020-12-08:
The isEOF() method tells you when you have reached the end of the file.

From the library documentation:

> Check if SMF is and end of file
> 
> An SMF is at EOF when there is nothing left to play (ie, all tracks have finished). The method will return EOF true when this condition is reached.
> 
> Returns
> true if the SMF is at EOF, false otherwise.
> 

The use is illustrated in ALL the examples, as they all need to know when the file is finished.

Please take some time to at least scan the documentation to understand what each function does.  The doc is in the library folder under docs - open the index.html file.

closed 2021-03-23 by MajicDesigns