MajicDesigns/MD_MIDIFile #17: PlatformIO only fetches tagged version v2.5 (not the newer v2.5.1)

philj404, 2021-03-22:
Thank you for creating this library.  I'm trying to get it working -- and I am getting closer -- but I am running into problems under PlatformIO.

First of all, _are you expecting MD_MIDIFile to work with PlatformIO?_  If not, you can just mark this issue as "WILL NOT FIX".

**If you would like I can ask the PlatformIO people to help fix this.**  It is not the first time this has happened, and I am sure it will not be the last.

I had a similar issue with my library.  I think PlatformIO has trouble changing default branch names (like you I changed my base branch from "master" to "main").  See https://community.platformio.org/t/latest-versions-of-simpleserialshell-lib-are-not-showed-in-platformio-repos/19387 (changes were in https://github.com/philj404/SimpleSerialShell/pull/7)

## Subject of the issue
The symptom I see is exactly the same as the one which should have fixed issue #15 -- that is, I get the compile time error because the SdFat library changed its public interface.  I see you made changes to address this in MD_MIDIFile for the v2.5.1 release.  It looks like PlatformIO only installs v2.5, and does not find the later version.

## Your Environment
VS Code 1.54.3/PlatformIO core: 5.1.1 home: 3.3.4

**Library Version:** "default version" -- v2.5
**Arduino IDE version:**  (see PlatformIO)
**Hardware model/type:** Arduino UNO
**OS and Version:** Windows 10

## Steps to Reproduce
1. Create a "MD_Midifile_Player_CLI" project for Arduino Uno.
2. Take the [MD_Midifile_Player_CLI.ino example](https://github.com/MajicDesigns/MD_MIDIFile/blob/main/examples/MD_MIDIFile_Player_CLI/MD_MIDIFile_Player_CLI.ino) and paste it into src/main.cpp.  Add "#include <Arduino.h>" to the top of the file.
3. Add the latest (greiman) SDFat library to the project.
4. Add the latest (majicdesigns) MD_MIDIFile library to the project.
5. Set your workspace to build your project and build.

## Expected Behaviour
The build should be successful. SDFat v2.0.4 and MD_MIDIFile v2.5.1 should be compatible with each other (at this time).

## Actual Behaviour
Instead I get a build error just as described in #15.

Looking at the libraries that are pulled in it looks like I _do_ get the latest SDFat version, but I only get MD_MIDIFile v2.5.

I can fix the problem by explicitly rolling back the versions of both libraries in platform.ini:

```` platformio.ini
...
; this lets you build, but does not let you pick up bugfixes for either library
lib_deps = 
	greiman/SdFat @ ~1.1.4
	majicdesigns/MD_MIDIFile@^2.5.0
...
````
No matter what, PlatformIO cannot pull in MD_MIDIFile 2.5.1.

MajicDesigns, 2021-03-22:
I think all my libraries work with PlatformIO - I get no complaints, in any case :) - so it would be consistent if this one does as well.

I don't use PlatformIO or know much about it, so if someone from the community wants to tell me they think needs to be fixed, I'll look at it.

philj404, 2021-03-22:
The fix _may_ be all on the PlatformIO side -- it was that way for my library.  I can make the request -- or at least confirm it is fixed -- if you wish.

I'm not really sure how PlatformIO picked up your library.  Maybe the PlatformIO library indexer iterates over github repositories (so it finds interesting code), but it doesn't completely understand/support an Arduino library's release structure.

I thought my [SimpleSerialShell](https://github.com/philj404/SimpleSerialShell) library was picked up because I published it for the Arduino platform.  I requested the Arduino team add it to "Manage libraries...".  But I do not see MD_MIDIFile is available through "Manage libraries...", so I suspect you have not [published the library](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) like that.

MajicDesigns, 2021-03-22:
PlatformIO seems to scan all the Arduino libraries that are in the Managed Libraries list. MD_MIDIFile used to be on the list but now seems to have dropped off the list. That could be an issue for PlatformIO as well. I'll raise a ticket on the Arduino site and see what they say..

MajicDesigns, 2021-03-23:
Back on the list for them to include it in the Managed Libraries. Apparently they have no record of it ever being listed. It should be back in a few days.

MajicDesigns, 2021-04-03:
The library is now listed. Does this make a difference to PlatformIO?

philj404, 2021-04-04:
I see the recent MD_Midifile v2.5.1 in the Arduino IDE.  Thanks!

So far I can only see the older MD_Midifile v2.5 picked up in PlatformIO.  

(At this point PlatformIO will not pick up v2.5.1 even when I explicitly tell it to look.)

**Now that you have the Arduino library manager behaving correctly (again), I will ask the PlatformIO folks to resynchronize their connection.**

By the way, thanks for your writing your 

- [three ](https://arduinoplusplus.wordpress.com/2018/05/11/playing-midi-files-on-arduino-part-1-standard-midi-files/)
-  [part](https://arduinoplusplus.wordpress.com/2018/05/24/playing-midi-files-on-arduino-part-2-keeping-to-the-beat/) 
- [article](https://arduinoplusplus.wordpress.com/2018/06/17/playing-midi-files-on-arduino-part-3-hardware-and-md_midifile-library/) 

about playing MIDI files with Arduino! That is how I discovered your library, even when the Library Manager lost track of it.

philj404, 2021-04-04:
I've [submitted a request](https://community.platformio.org/t/md-midifile-library-is-stuck-on-version-2-5-v2-5-1-is-missing/20477) asking for help synchronizing Arduino library updates on PlatformIO.

I will let you know how things turn out.

philj404, 2021-04-05:
PlatformIO now shows v2.5.1 as expected.  Thanks for your help!

closed 2021-04-05 by philj404