MajicDesigns/MD_MIDIFile #18: Issue in Playing from SDcard in STM32F1 Bluepill Maple core

Yousaf93, 2021-05-19:
## IMPORTANT
Before submitting this issue
[ ] Have you tried using the latest version of the library?
Ys
[ ] Have you checked this has not already been submitted and/or resolved?
Yes
[ ] If you are requesting help a better choice may be the [ Arduino forum](http://forum.arduino.cc/)
I have tried there too

## Subject of the issue
Issue in Playing from SDcard in STM32F1 Bluepill Maple core

## Your Environment
**Library Version:**  Latest
**Arduino IDE version:** 1.8.13
**Hardware model/type:** STM32 Blue pill
**OS and Version:**  Maple Core

## Steps to Reproduce
I am trying to play a file in loop. 
## Expected Behaviour
In Arduino Uno i was able to play a single file continuesly using MD file play example but it just plays once in bluepill

## Actual Behaviour

## Code Demonstrating the Issue
````
// Test playing a succession of MIDI files from the SD card.
// Example program to demonstrate the use of the MIDFile library
// Just for fun light up a LED in time to the music.
//
// Hardware required:
//  SD card interface - change SD_SELECT for SPI comms
//  3 LEDs (optional) - to display current status and beat. 
//  Change pin definitions for specific hardware setup - defined below.
//#include <Adafruit_DotStar.h>
#include <SdFat.h>
#include "sdios.h"
#include <MD_MIDIFile.h>
//#include <SPI.h>  
#include <FastLED.h>
#define NUM_LEDS 61
#define USE_MIDI            0       // set to 1 to enable MIDI output, otherwise debug output
#define DATA_PIN PB1
#define CLOCK_PIN PB0
//#define NUMPIXELS          61    // Popular NeoPixel ring size
//Adafruit_DotStar strip(NUMPIXELS, LED_DATA_PIN, LED_CLK_PIN, DOTSTAR_BRG);
CRGB leds[NUM_LEDS];
#if USE_MIDI // set up for direct MIDI serial output

#define DEBUG(x)
#define DEBUGX(x)
#define DEBUGS(s)
#define SERIAL_RATE 9600

#else // don't use MIDI to allow printing debug statements

#define DEBUG(x)  Serial.print(x)
#define DEBUGX(x) Serial.print(x, HEX)
#define DEBUGS(s) Serial.print(F(s))
#define SERIAL_RATE 9600

#endif // USE_MIDI
#define LED_PIN PC13
//#define BUTTON_PIN 38
#define LED_RATE 500000  
void handler_led(void);
int toggle = 0;
// SD chip select pin for SPI comms.
// Arduino Ethernet shield, pin 4.
[... 243 lines cut ...]
    midiSilence();
    timeStart = millis();
    state = S_WAIT_BETWEEN;
    DEBUGS("\nWAIT_BETWEEN");
    break;

  case S_WAIT_BETWEEN:    // signal finished with a dignified pause
    //digitalWrite(READY_LED, HIGH);
    //strip.setPixelColor(READY_LED, 0x00ff00);
    //strip.show();
    if (millis() - timeStart >= WAIT_DELAY)
      state = S_IDLE;
    break;

  default:
    state = S_IDLE;
    break;
  }
}

void handler_led(void) {
    toggle ^= 1;
    digitalWrite(LED_PIN, toggle);
} 
````

MajicDesigns, 2021-05-20:
I don't have your processor hardware configuration, so I can only make suggestions that you need to try.
1. Do you get an debug output that says there is an error anywhere? 
2. You are using FastLED. This library is really heavy on the interrupts and could cause issues if it .show() is called a lot. In general these functions should only be called when there is a color change, not every time through lopp() for example. Maybe try running with the FastLED.show() code disabled.
3. Try not calling midiSilence(). Maybe (a long shot) the MIDI output is being turned off and you can't hear the music playing, but this would have been the same with the Uno.

Yousaf93, 2021-05-20:
Hi Marco , Thanks for reaching out.
1)No, the debug output just displays the last notes played and their state
and nothing after that.
2)FASTLED.show is only called when I receive a note from the MIDI file and
it was working absolutely fine with the Arduino UNO in the same application.
The only change is the SDfat object as you can see in the example the
object made from a different class and I am using a different one because
of compatibility with STM32. I used a timer blink to check if the
controller is stuck at some place but the blink always goes on so that
means the controller is running but its not able to repeat the track somehow
Regards
Yousaf Masood

MajicDesigns, 2021-05-21:
[quote]
1)No, the debug output just displays the last notes played and their state and nothing after that.
[/quote]
It cannot be 'nothing' as at least the state name is printed (eg, S_IDLE, S_WAIT_BETWEEN) when the state code is executed. You have a Finite State Machine controlling the execution of the playing, so you should be able to work out what state this is stuck in and home in on what could be preventing it from moving on.

Yousaf93, 2021-05-21:
Yes i tried to print the test statements in each of the states but was
amused that nothing got printed after  the last callback prints

MajicDesigns, 2021-05-21:
So it is probably stuck in WAIT BETWEEN.

Yousaf93, 2021-05-21:
Ok Thanks alot.
I will look into it and get back to you with updates

closed 2021-05-31 by MajicDesigns