MajicDesigns/MD_MIDIFile #19: Adaptation onto ESP32 Dev module

VividHobby, 2022-01-02:
Dear owner,
Thank you very much for this usefull MIDI application.
I an trying to adapt this onto ESP32 after successful test working on ArduinoNano, but I have an ERR
at openning SD card.
Is there any additiona settings when we adapt this "MD_MIDIFile" onto ESP32 Dev module ?
There was only one point I did was new "SPI CS (pin #5) for VSPI on ESP32.
Thsnk you for your advices,

Best Regards
Nobuaki

MajicDesigns, 2022-01-02:
I don't think any changes are needed to run any of the library example programs this library on ESP-32. All the SD card operations are using the SDFat library.

As a first step you should verify that the SD card library (SDFat) works properly, using one of the example programs from that library.

VividHobby, 2022-01-02:
Dear Marco
Thank you very much for quick reply on my previous message.And thank you for advice.
I alread have done trying this on Arduino Nano, and result was OK.Beat LED was blinking beat b beat.
And then, I modified followings
1) Chane SS (CS) from Pin# 10 to 52) change Arduino Nano SPI (15, 16, 17) to ESP32 VSPI( 23,19, 18)      on my breadboard. (no assignment on the scketch)
Compile was done successfully, but the scech output "SD init fail".
Any advise ?
Thank you for your help.

BestReagrdsNobuaki 

     ----- Original Message -----

MajicDesigns, 2022-01-02:
No, you did not test as I suggested. Please test SDFat with the SDFat examples on the Target hardware. If SDFat is not working there is no way the midi library will work.

VividHobby, 2022-01-02:
Oh !I see, thank you for your kindness.I will try it out.
RegardsNobuaki

     ----- Original Message -----

closed 2022-04-07 by MajicDesigns