MajicDesigns/MD_MIDIFile #20: implanter un encodeur rotatif

music59940, 2022-03-20:
Bonjour

Je souhaiterais intègre un encodeur rotatif pour générer des ticks.

Je souhaiterais faire comme un orgue de barbarie.

quand on tourne la manivelle ca génère le tempo

merci pour votre réponse

MajicDesigns, 2022-03-21:
I am not sure there is a question in your request. What you propose should be possible but is there something specific that you are asking?

music59940, 2022-03-21:
Merci pour votre réponse.
j'aimerais utilisé l'exemple MIDIFILE_LCD.
On sélectionne le morceaux, le tempo est a 0 si on tourne pas la manivelle.
après quand on tourne la manivelle on augmente ou diminue le tempo en fonction de la vitesse de rotation de la manivelle.

j'ai essayer plusieurs fois de modifier le code mais j'y arrive pas.
merci pour votre aide.

(l'encodeur rotatif donne 20 impulsion par tour)

MajicDesigns, 2022-03-22:
I don't think you are starting from the right place. In order to play the music only when you are turning the crank, you need to generate the 'ticks' for the library rather than rely on it to keep to the 'correct' time.

The Tempo example shows how you do this, so please understand how that example works. Yu need to make sure that GENERATE_TICKS is set to 1.
````
// Set to 1 to generate the MIDI ticks in this application
// Set to 0 to allow library to generate clock based on Tempo
#define GENERATE_TICKS  1
````

music59940, 2022-03-22:
Il y as un bug dans l'exemple "play lcd"
MD_MIDIFile_Play_LCD:105:28: error: macro "DEBUG" passed 2 arguments, but takes just 1
   DEBUG("\nM T", pev->track);
                            ^
MD_MIDIFile_Play_LCD:106:33: error: macro "DEBUG" passed 2 arguments, but takes just 1
   DEBUG(":  Ch ", pev->channel+1);
                                 ^
MD_MIDIFile_Play_LCD:109:29: error: macro "DEBUGX" passed 2 arguments, but takes just 1
     DEBUGX(" ", pev->data[i]);
                             ^
MD_MIDIFile_Play_LCD:118:28: error: macro "DEBUG" passed 2 arguments, but takes just 1
   DEBUG("\nS T", pev->track);
                            ^
MD_MIDIFile_Play_LCD:121:29: error: macro "DEBUGX" passed 2 arguments, but takes just 1
     DEBUGX(" ", pev->data[i]);
                             ^
MD_MIDIFile_Play_LCD:160:26: error: macro "DEBUG" passed 2 arguments, but takes just 1
   DEBUG("\nLCDErr: ", msg);
                          ^
MD_MIDIFile_Play_LCD:188:29: error: macro "DEBUG" passed 2 arguments, but takes just 1
       DEBUG("\nFile ", count);
                             ^
MD_MIDIFile_Play_LCD:189:23: error: macro "DEBUG" passed 2 arguments, but takes just 1
       DEBUG(" ", fname);

un autre bug dans l'exemple "tempo"
des que le tempo est au dessus de 136 
le tempo reviens a 0

MajicDesigns, 2022-03-23:
v2.5.3 corrects all the errors from DBUG statements (changed recently).

I don't see what can be causing the error with BPM > 136. Maybe you can track when it is happening and give more deuils.

music59940, 2022-03-24:
merci d'avoir corriger les bugs

je voudrais combiner "exemple tempo" avec "exemple play lcd"
mais je n'y arrive pas
merci pour votre aide

MajicDesigns, 2022-03-25:
It is definitely possible to do this. You need to focus on code in the midiFSM() function at

````
  case MSProcess:
    // Play the MIDI file
    if (!SMF.isEOF())
    {
      if (SMF.getNextEvent())
      {
        char  sBuf[10];
        
        sprintf(sBuf, "T%3d", SMF.getTempo());
        LCDMessage(0, LCD_COLS-strlen(sBuf), sBuf, true);
        sprintf(sBuf, "S%d/%d", SMF.getTimeSignature()>>8, SMF.getTimeSignature() & 0xf);
        LCDMessage(1, LCD_COLS-strlen(sBuf), sBuf, true);
      };
    }    
    else
      s = MSClose;
````
This is currently playing the musing with the tempo from the music. You need to change this to play with your tempo from the rotary encoder, as per the other example.

If you need more help than this then you should move the conversation to the Arduino forum (arduino.cc). This forum issues list is more for bug reporting or very short 'help' questions. Let me know if you do this so I can continue there.

music59940, 2022-03-28:
Bonjour

désolé de répondre aussi tard car j'été en déplacement.
je vais essayer cela et j'irais sur le forum Arduino si je n'y arrive pas.
merci beaucoup.

closed 2022-03-28 by MajicDesigns