MajicDesigns/MD_MIDIFile #25: Compile and runtime problems with Examples

ElbyMann, 2022-07-08:
I'm trying to get examples running on an Arduino Uno, and am stumbling over two issues.  I'm using MD_MIDIFIle 2.5.3 with SDFat 2.1.2, both installed via library manager on IDE 1.8.19.

1)  Trying to compile MD_MIDIFile_Dump, I get the following compile error:

````
C:\Users\Elby\AppData\Local\Temp\ccHX91ye.ltrans0.ltrans.o: In function `loop':
C:\Users\Elby\Documents\Arduino\libraries\MD_MIDIFile\examples\MD_MIDIFile_Dump/MD_MIDIFile_Dump.ino:89: undefined reference to `MD_MIDIFile::dump()'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Uno.

````
  
Sketchbook location is: C:\Users\Elby\Documents\Arduino and I've verified that "C:\Users\Elby\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDIFile.h" exists.  I can see a "dump" method within when I open this file.   If I rename the file, the compile error becomes a  "missing .h file" notification, which seems to indicate that originally, the compiler is finding the file.

2)  Trying to run MD_MIDIFile_Play, with the following changes: 
    - comment out USE_MIDI at line 13
    - set SD_SELECT = 4 at line 36
   
 I get the following messages in the serial monitor:
````
[MidiFile Play List]
SD init fail!
````
I've verified that SD_SELECT = 4 is correct by making the same change to line 13 of MD_MIDIFIle_Dump_Raw, and seeing the SD directory listed correctly in the serial monitor.

Please let me know if you need additional information.  I'd really like to get these examples running, so I can start development; thanks for your help!

MajicDesigns, 2022-07-08:
From the top of the MD_MIDIFile_Example.ino file:
`
// MIDIFile.h library settings must be set to values below before compiling
// #define  DUMP_DATA         1
// #define  SHOW_UNUSED_META  1
//
// -= REMEMBER =-
//  * Turn line ending in Serial Monitor to NEWLINE
//  * Turn DUMP_DATA and SHOW_METADATA after back to default (0) when done
//
`
Edit the MD_MIDIFile.h file to turn on these switches (which are ONLY for debugging) and remember to switch back when you want to create production code.

The INIT FAIL error comes form the SDFat library so you will need to test the SD card setup with the code from that library (ie, make sure that they work before using the MIDI file library).

closed 2022-07-08 by MajicDesigns

ElbyMann, 2022-07-09:
Thanks, Marco!  Loving MD_MIDIFile!! 😊