MajicDesigns/MD_MIDIFile #28: Use of function min does not have strictly typed parameters.

markeby, 2024-08-19:
The use of the standard function "min" does not have strictly type parameters that match.  This causes conflicts with the latest version of the ESP32 standard library and conflicts with the use in the STL library. 

This is for version 2.4.2 of the Arduino IDE and version 1.0.2 of arduino-cli.

Compiling with the latest libraries has errors due to this issue.

tallherve, 2024-12-07:
Hi, any update on this? Cannot compile for an ESP32-C3 as this chip is not included in ESP library version1.02.
I downgraded to 2.0.17 but I have the first error as

MajicDesigns, 2024-12-07:
The libraries should be updated with markby commit that supposedly fix the issue (untested by me but the look ok).

closed 2024-12-07 by MajicDesigns

tallherve, 2024-12-08:
Still not compiling for me with ESP32 3.1.0R , either for a ESP32 dev board (this error message: 
"thread 'main' panicked at 'assertion failed: `(left != right)`
  left: `0`,
 right: `0`: Failed to get path name. Error code: 3', main.rs:65:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
exit status 101

Compilation error: exit status 101"
)

or an ESP32-C3 (starting with the previously passed error message with min function).

No problem with an Arduino NANO.

MajicDesigns, 2024-12-09:
That looks like a different problem now. Does the compiler tell you what file/line number it is objecting to?

tallherve, 2024-12-09:
Using ESP32 v2.0.7, it compile for an ESP32 dev module but for an ESP32C3 Dev Module, it gives the following errors:
(line 417 seems to create the problem, I understood that the first message was about this issue too. I see that this line 417 was modified by Markeby).
I'll try again with ESP32 v 3.1.0R

> c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDITrack.cpp: In member function 'void MD_MFTrack::parseEvent(MD_MIDIFile*)':

c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDITrack.cpp:417:56: error: no matching function for call to 'min(unsigned int, uint32_t&)'
         uint8_t minLen = min(ARRAY_SIZE(mev.data), mLen);
                                                        ^
In file included from c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\algorithm:62,
                 from C:\Users\fonta\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\cores\esp32/Arduino.h:172,
                 from c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDIFile.h:385,
                 from c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDITrack.cpp:23:
c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3456:5: note: candidate: 'template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)'
     min(initializer_list<_Tp> __l, _Compare __comp)
     ^~~
c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3456:5: note:   template argument deduction/substitution failed:
c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDITrack.cpp:417:56: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned int'
         uint8_t minLen = min(ARRAY_SIZE(mev.data), mLen);
                                                        ^
In file included from c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\algorithm:62,
                 from C:\Users\fonta\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\cores\esp32/Arduino.h:172,
                 from c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDIFile.h:385,
                 from c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDITrack.cpp:23:
c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3450:5: note: candidate: 'template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)'
     min(initializer_list<_Tp> __l)
     ^~~
c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3450:5: note:   template argument deduction/substitution failed:
c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDITrack.cpp:417:56: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned int'
         uint8_t minLen = min(ARRAY_SIZE(mev.data), mLen);
                                                        ^
In file included from c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\algorithm:61,
                 from C:\Users\fonta\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\cores\esp32/Arduino.h:172,
                 from c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDIFile.h:385,
                 from c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDITrack.cpp:23:
c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:243:5: note: candidate: 'template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)'
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^~~
c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:243:5: note:   template argument deduction/substitution failed:
c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDITrack.cpp:417:56: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned int' and 'uint32_t' {aka 'long unsigned int'})
         uint8_t minLen = min(ARRAY_SIZE(mev.data), mLen);
                                                        ^
In file included from c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\algorithm:61,
                 from C:\Users\fonta\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\cores\esp32/Arduino.h:172,
                 from c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDIFile.h:385,
                 from c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDITrack.cpp:23:
c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:195:5: note: candidate: 'template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)'
     min(const _Tp& __a, const _Tp& __b)
[... cut ...]
c:\users\fonta\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:195:5: note:   template argument deduction/substitution failed:
c:\Users\fonta\Documents\Arduino\libraries\MD_MIDIFile\src\MD_MIDITrack.cpp:417:56: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned int' and 'uint32_t' {aka 'long unsigned int'})
         uint8_t minLen = min(ARRAY_SIZE(mev.data), mLen);
                                                        ^

exit status 1

Compilation error: exit status

tallherve, 2024-12-09:
With ESP32.1.0R, it does not compile for an ESP32 Dev Module, with the same error on line 417. I can pass the full message if necessary.

MajicDesigns, 2024-12-09:
The line 417 displayed by the compiler seems to be from the unmodified library. Can you verify that you are using the latest? Check line 417 in GitHub against the one in your copy of the library and make sure you don’t have 2 copies of the library folder somehow.

tallherve, 2024-12-10:
Thanks a thousand.
I only have one library MD_MIDIFILE, this is 2.6.0, installed through the Arduino IDE. However all the lines changed by markeby were without the modifications.
I changed them as per the commit 8212de2 and voila... all compiled ok.
That was my first install of this library so expected it to be the latest one, I have no idea how this happened.
Thanks again for the help.