MajicDesigns/MD_MIDIFile #31: Can't control drum machine in time

lazyoldjack, 2025-05-09:
I have a midi file that I can play to a DR-550 drum machine from the Windows computer and it plays fine, each beat perfectly in time.

Based on the examples on this github, using the library, I have created a "MIDI player" using a "Cheap Yellow Display board" (or CYD as they are known) which has a screen and an ESP32,  

When I try to play the same file with my program  on the CYD (via the UART) to the DR-550, it nearly gets on the beat at the start, but gradually gets more and more off the beat. If I try to play a REALLY simple file, it can play it OK. 

I can think of a number of possibilities for the problem (other RTOS tasks interfering with timing, some problem with the fact that the UART is also connected to the USB port chip, LVGL graphics library being too hungry for processing) but I have gone to the ends of the earth to try to mitigate these aspects (tasks minimised, UART instructed to ignore start stop signalling, LVGL tick only hit when not playing, etc. etc)..

I can, and will if useful, post the (now very messy multiple-times-altered) code, but I thought that you might have a few pointers to things you've discovered before that can make all the difference? Even knowing that no one has ever had this problem before would be useful!!

MajicDesigns, 2025-05-10:
This problem has not been reported before. That does not mean that it has not occurred, just that it has not been relayed as an issue.

The MIDI file player tries to keep to the beat as best it can, bearing in mind that it uses millis() to count out elapsed time and it does not synchronize to the MIDI time signal (if there is one). It is designed to play a self-contained MIDI score consistently, but you may expect that it will get out of synch with other scores being played by other instruments over time.

My only suggestion is the obvious. If you can, to try and play the piece without the complications of the RTOS, etc, and see if it keeps time. Maybe just play it on a simple processor like an Uno. This should tell you if the problem lies in the library coping with the MIDI score complexity.

If it does not work in this simplest form, then this is not the right library for you. 

If it does, then re-introduce complications until it stops working and you should have an indicator of the cause.

lazyoldjack, 2025-05-10:
Thanks for the info that it has not been reported before (I certainly couldn't find it it your issues). Reading again what I wrote above and your excellent reply has given me another idea... just going to check...

OK, so I'm back, very excited because it's working!! Talking with you made me wonder if the speed restriction was elsewhere in the system, and sure enough, I had missed a 0 off the SD card speed **_(DOH!)_** and now it works like a dream! **Thank you so much for your help and your excellent library.**

closed 2025-05-10 by lazyoldjack